1V-Oct/3318_the_centre_releases #10: Update bricked my module.

Pipstapee, 2021-06-05:
Update bricked my module. I'm going to have to reflash the firmware with the previous. We couldn't get the dump to work and the previous update didn't work via the SD if you remember, you sent me the update in the form of a complete firmware HEX.

Pipstapee, 2021-06-05:
[image]

duddie, 2021-06-05:
No, it didn't brick it :)

What has happened is that it right now has some extra overlays in the firmware FWX file and this file has to be present on the SD card all the time.

So you need the_centre_v3.fwx on the SD crad.

Pipstapee, 2021-06-05:
And rename it?
That's what caused it.
I've reflashed it with the HEX you gave me for the previous update.

Pipstapee, 2021-06-05:
Current FW. 0.61.40, doesn't update with it named the_centre_v3 but if I rename it it results in the above photo...

duddie, 2021-06-05:
Please download the_centre_v3.fwx and put the file twice on SD-card. Once name it the_centre.fwx and once with original name. It should work. The file the_centre_v3.fwx must remain on the SD Card inside SD card slot all the time. Some units (like your) are older version without extra SPI memory which means the firmware must be loaded from card all the time.

So again, keep two files. Same version.

duddie, 2021-06-05:
As discussed. Problem solved :) Closing.

closed 2021-06-05 by duddie