1V-Oct/3318_the_centre_releases #13: Unexpected Patch Reset

CloneCrow, 2021-10-04:
Loving the module...something that keeps happening below...

I have a sequence incoming at V/OCT1.
Add a slow changing LFO to change the waveshape.
Back to the wave table, press input - the patch occasionally rests to default.

Lots of great potential in this module. -:)

duddie, 2021-10-06:
Hi,

Can you please upgrade to the latest firmware, save patch to SD card, reset module, load patch and from that point try to reproduce and give me all steps in order. Also please attach patch to the Issue Ticket (here) and if necessary the required wavetable (please zip it to not let GitHub recompress it). I tried many times and cannot reproduce.

CloneCrow, 2021-10-06:
Hi There,

After I saved my patch it didn't do it so often, I had 3 wave table
running. I will video and document as I go so I have all the steps for you.

Many thanks for all you have done for us.

Terry

CloneCrow, 2021-10-06:
> If it does it however not often please try to capture it. If I can
> reproduce it I can fix it.
>
> If you can reproduce the crashes that you have done before saving then
> please document and send me with all files. Lets fix easy crashes first.

Hi there,

Well, I loaded the latest firmware and the issue seems to have gone away
for the time being...not sure what's happened but it seems more stable in
that respect.

I've not been able to make it do it again...so fingers crossed.

Many thanks for all your help and attention.

Terry

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>