1V-Oct/3318_the_centre_releases #16: [REQUEST] Please add folder nesting to file tree

thegreenalien, 2021-10-05:
At present, file system doesn't allow separate folders within categories, so all wavetables, for example, need to be stored together in the 'tables' category. This will make finding specific wavetables increasingly difficult as more are added.
Please add the option to group items within folders.

ExitChamber, 2021-10-05:
Mine is working fine with wavetables in separate folders within the "tables" directory

[image]

thegreenalien, 2021-10-05:
Very interesting. I tried several times, but the files were only visible in the base folders. I'll try again.

ExitChamber, 2021-10-05:
How are you trying to load them? I was a bit confused at first until I realised that you need to select your wavetable by adding a WTO and then clicking the encoder at the wave display screen:

[image]

That should bring up the file structure as shown in that last pic.

thegreenalien, 2021-10-05:
I got exactly to that stage, clicking the encoder just brought up a blank
screen. No problem loading wavetables when they are placed directly in the
'tables' directory.

ExitChamber, 2021-10-05:
Oh, that's very odd. Sorry I'm out of suggestions.

thegreenalien, 2021-10-05:
Funny, it's working fine now, no idea why ir wasn't working yesterday! Thanks for the feedback, issue closed.

closed 2021-10-05 by thegreenalien