1V-Oct/3318_the_centre_releases #27: V/OCT latency when using external envelope.

lookoutjames, 2021-10-11:
I'm seeing very long (~35ms) latency when using The Centre purely as an oscillator, triggering via an external envelope. 

My setup is as follows:
- NiftyCASE MIDI interface taking MIDI input from an external keyboard
- The Centre, with a simple patch (same effect with either wavetable or VCO) using V/OCT1 input to control pitch (from MIDI CV1) and sending output to VOUT1
- PipSlope triggering a short-attack short-delay gated envelope from MIDI GATE1 (so effectively 'on' when a key is pressed)
- VOUT1 going via a 2hp VCA modulated using the output of PipSlope

The effect is that the note is 'on' precisely when I press the key, but the pitch changes are always noticeably late. I've attached a .wav file (in a zip file) showing the effect. 
[3-Audio_0001_2021-10-10_124957.zip](https://github.com/1V-Oct/3318_the_centre_releases/files/7324016/3-Audio_0001_2021-10-10_124957.zip)

If I use the internal envelope generator/VCA (e.g. using the 'Simple Preset WTO + ENV + VCA' patch from the discord) with MIDI CV1 into V/OCT1 and MIDI GATE1 into CVY1 I don't hear the problem (which might imply that the CVY inputs have the same latency response as the V/OCT ones).

Ideally I wouldn't be able to hear any delayed note change when using The Centre as an oscillator with an external envelope: that's my main use case, really: I'm interested in the oscillator editing capabilities but less on constructing complex internal patches using the internal envelopes. It is possible to delay external envelopes by a certain amount but that requires extra modules and I'd rather not have to go down that route if I can help it.

lookoutjames, 2021-10-12:
I've just had a play and can confirm that setting 'Exp CV Scan' to On helps things a lot. It's still not quite perfect: with notes spaced far apart there seems to be almost a fast slide between them, but things are a lot better. I couldn't detect any difference when I fiddled with any of the other three parameters. I'll do some more checking later on.

thegreenalien, 2021-10-13:
I am experiencing similar problems, but with a different midi to cv interface. (Erica Synths) I wonder how fast the midi to cv conversion is, trying a voltage meter on my converter does show a noticable short delay as the voltage ramps up, but some of this delay may be due to the meter display lag. I'll try it with an oscilloscope and report back.

lookoutjames, 2021-10-13:
@duddie: OK, I've tried all of these. I couldn't tell any difference between 1, 2, and 3 (they all sound pretty good, with an audible but tiny latency), or between 4 and 5 (which both exhibit the original problem). 

One thing I did notice is that if I do a test with the pitch bend wheel on my keyboard, I only get the semitones (and the same happens if I put e.g. Maths channel 3 into V/OCT and turn the knob slowly). Is the V/OCT input quantising the signal? Can that be turned off, if we want properly swoopy patches, and could that be interfering with this latency issue? 

@thegreenalien: my MIDI interface doesn't have a glide feature; the issue I was having wasn't to do with a slow change in V/OCT level, it was because The Centre is reacting late to an instantaneous change in V/OCT level. When I use other (digital) oscillator modules (e.g. Plaits, or a Disting) in the same way, there's no latency.

duddie, 2021-10-14:
Yes, The Centre quantises everything because of this issue. It's a hardware limitation that I mentioned earlier - very noisy ADC. So the solution is to accept that the pitch control will be gliding and shaking before it stabilises because of the noise.

Other solution that in my opinion is better is to add Frequency Fine / Frequency Coarse control to WTO/VCO like LFO and use CV input.

Maybe let's have proper discussion about it

rockstarartist, 2021-10-15:
This is interesting. I havent played with external envelopes with the Centre yet. I am going to give this a try.
I will try this out with my  Expert Sleepers FH-2 as my MIDI input to CV out, an Joranalogue Contour 1 for envelope, and i too have a 2HP VCA.

rockstarartist, 2021-10-15:
Okay, so in 'The Centre' version 0.73.38 i couldnt discern a noticeable difference in lag when I tried this.

Was this corrected in version 0.73.38?

Midi keyboard into FH-2 via USB
FH-2 pitch into the centre (simple patch, 1x WTO into outs) via V/OCT 1
FH-2 gate into Joranalogue Contour 1, Fast attack, semi-fast decay envelope trigger
Joranalogue Envelope out into 2hp VCA CV input
The Centre Audio out into 2hp VCA CV input.

duddie, 2021-10-16:
No, there are two problems:
1. This chip adds a lot of noise when the driver for LCD does updates. The ADC (analog to digital converters) are very noisy at this time. Since the update takes like 11 ms I have to alternate reading ADC and Screen updates. That's the lag introduced there. I can make a version with ability to turn off screen and test it but that is not a solution. The better solution is to interleave screen updates into a slices, but it's a bigger rework that I can do maybe next month (unless this is a bigger problem and needs urgent action)
2. Those ADCs are noisy anyway by themselves so I am applying certain 1-pole filtering there and it adds this particular glide. This would not be a problem if the screen updates were not delaying it. So if I can fix screen updating mechanism I should eliminate rest of the lag.
3.

rockstarartist, 2021-10-16:
> @rockstarartist: your setup sounds functionally identical to mine, but I was definitely seeing a lag with that that I didn't see when I put the MIDI pitch into different oscillator modules: I don't have a Mordax (on my xmas list...) so I don't have a trace, but I put the audio clip of the effect on the issue.
> 
> The new settings that made it much better for me were added in 0.74.11 (with the third 'Settings' item in the system menu), so if you're before that version I guess you're just not seeing the problem at all. @duddie, could the latency be hardware-specific? I know you said you weren't very impressed with the noisy ADC: could some hardware produce worse delays?
> 
> In fact, having experimented with some of the internal envelopes, I can still detect a little latency even if it's using its own envelope to trigger from (i.e. taking a gate input on CVY1 to trigger an internal envelope for an oscillator coming in on V/OCT1).

Ah interesting, so you are saying, given two or more WTO modules in the centre, reading from the same pitch source you are getting that variance as shown in your audio file? I was only testing with one WTO module. I'll give it a try tonight. I am on a later version now though, so I am guessing my experience will be different if there have been corrections already in place.