1V-Oct/3318_the_centre_releases #31: [BUG] Adding waveform to second WTO crashes

thegreenalien, 2021-10-13:
With one WTO up and running with waveform loaded, a second WTO is added. When the waveform for this WTO is selected, the unit crashes.
To test this I tried loading the same waveform (Producer hive SqFifth1) into both WTO, it crashed.
Tried with several waveforms, still crashes when second one is added.

duddie, 2021-10-14:
Please see the private message on Discord. 
I have tried reproducing. I have 3 WTOs with different or same wavetables. Cannot reproduce.

duddie, 2021-10-14:
I spent like 2 hours in different combinations tryin to reproduce this problem to not experiencing a single crash. If you can give me all details it will be much easier. Please mail me WAV files used (hopefuly you can recreate it with single wav). I assume you are using the latest firmware.

thegreenalien, 2021-10-15:
> I'm going to try creating a new system disc with just one set of tables on
> it and see what happens.

I've tried it with a new system disc that just contains one set of tables,
it works fine. I'll now add more sets of tables, see what happens.

thegreenalien, 2021-10-15:
I've found some wavetables that definitely cause problems, I've emailed you
a couple of zip files.

duddie, 2021-10-16:
How is the problem now?

duddie, 2021-10-31:
I hope all is good now?

closed 2022-09-19 by duddie