1V-Oct/3318_the_centre_releases #46: Encoder reverse direction setting not the same from main menu to sub-menus

markcrosbie, 2022-03-28:
I'm on the 1.4.18 firmware. 

Encoder direction is set to 'off'. Turning the encoder in the main patch view results in a different direction of scrolling than in the sub-menus. They are reversed.

celichtenstein, 2022-05-29:
Thank you for reporting this. I'm experiencing the same issue.

The main menu to select a component to `Add` to the patch scrolls normally, but as soon as I enter the `Inputs` menu, the scroll direction is backwards and confusing.

duddie, 2022-07-29:
Please retest on 1.99 and let me know your thoughts.

celichtenstein, 2022-09-19:
Hi @duddie !

Just tested this again using the latest 2.0 firmware and I'm still having some issues.

✅ It seems the scroll direction has been fixed for the main patch view and when scrolling through the parameter groups while editing a patch device's inputs (🙌🏼!)
❌ However, one level deeper when selecting a single parameter at a time, the scroll direction is still backwards.

Here's a video showing the scroll behavior. The backward scrolling starts at about 14 seconds in the video:

https://user-images.githubusercontent.com/4134679/190942089-b93906db-a335-4cec-b8ae-1affcb6fe637.mov

duddie, 2022-09-19:
OK, will fix. 

You are aware you do not need to get into individual settings now? The 4 components of inputs can be changed by using LVL1-4 and ATT1-4 to change value and input source respectively.

Please check user-manual.pdf :)

But yeah, I will patch it today.

duddie, 2022-10-12:
The latest version should have all encoder reverse issues resolved. Also there are 4 settings for reversing encoder now.

closed 2022-10-12 by duddie